arduino/ArduinoCore-nRF528x-mbedos #104: I2C scanner with repeated start isn't working for Arduino Nano 33 BLE

arpita-agrawal, 2020-09-17:
Hi

I wanted to create an I2C scanner with repeated start enabled for Arduino Nano 33 BLE. Not calling Wire.write between beginTransmission and endTransmission the code kind of hangs and after a long time shows I2C device not detected.

However, if repeated start condition is not used the devices are detected properly. This issue is seen only for mbed 1.1.4. The same code works well on AVR, SAM and SAMD.

Please note that, if I write/read data from a specific I2C device with repeated start condition the issue isn't seen.

Please refer to the sketch below to reproduce the issue. Also note that , if scanner code below is run with all instances of endTransmission(1)(disabling repeated start) the issue isn't seen:
// --------------------------------------

#include <Wire.h>

void setup() {
  Wire.begin();

  Serial.begin(9600);
  while (!Serial); // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}

void loop() {
  int nDevices = 0;

  Serial.println("Scanning...");

  for (byte address = 1; address < 126; ++address) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    byte error = Wire.endTransmission(0);

    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) {
        Serial.print("0");
      }
      Serial.print(address, HEX);
      Serial.println("  !");

      ++nDevices;
    } else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) {
        Serial.print("0");
      }
      Serial.println(address, HEX);
    }
  }
  // send stop condition for last address
  byte address = 126;
  Wire.beginTransmission(address);
  byte error = Wire.endTransmission(1);

  if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) {
        Serial.print("0");
      }
      Serial.print(126, HEX);
      Serial.println("  !");

      ++nDevices;
    } else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) {
        Serial.print("0");
      }
      Serial.println(126, HEX);
    }
  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  } else {
    Serial.println("done\n");
  }
  delay(5000); // Wait 5 seconds for next scan
}

This sketch is an adaptation of i2c_scanner.ino example shipping with AVR Wire library from Arduino

arpita-agrawal, 2020-09-29:
Any updates on this?

facchinm, 2020-09-30:
@arpita-agrawal when scanning you need to ask for an address and check if the remote device responds.
Changing address between tests means that the transmission should be stopped, otherwise the master will always send repeated starts to different slaves.
So, the right way to scan is by calling `Wire.endTransmission(true);` and just check for the return result.

closed 2020-09-30 by facchinm