arduino/ArduinoCore-nRF528x-mbedos #108: digitalWrite 0 from D0 and D1 on Nano 33 BLE Sense doesn't work. Pins are always high

arpita-agrawal, 2020-11-03:
digitalWrite 0 from D0 and D1 on Nano 33 BLE Sense doesn't work. Pins are always high. Verified this on 3 different boards

Please use the below sketch to reproduce the issue;

//Short D0 to D3 and D1 to D2
void setup() {
   Serial.begin(9600);
   pinMode(0,OUTPUT);
   pinMode(1, OUTPUT);
   pinMode(2, INPUT);
   pinMode(3,INPUT);
   bool d2Val = digitalRead(2);
   bool d3Val = digitalRead(3);

   Serial.print("D2 ");
   Serial.println(d2Val);
   Serial.print("D3");
   Serial.println(d3Val);
}

void loop() {
  digitalWrite(0,0);
  digitalWrite(1,0);
   bool d3Val = digitalRead(3);
   bool d2Val = digitalRead(2);

   Serial.print("D2 ");
   Serial.println(d2Val);
   Serial.print("D3 ");
   Serial.println(d3Val);

   delay(150);
}

facchinm, 2020-11-03:
Which core version are you using? I couldn't reproduce on 1.3.0 (the pins move as expected).
Please note that old cores had a bug so `digitalRead()` was enabling the internal pin pullup, so this could be your problem.
Can you update the core and report if it's still broken in your setup? Thanks

arpita-agrawal, 2020-11-03:
Thanks, Martino for a quick response. The issue got resolved with Mbed OS version 1.3.0.

closed 2020-11-19 by facchinm