arduino/ArduinoCore-nRF528x-mbedos #112: Serial communication between ATMega 32U4 and Arduino nano 33 Iot

Mimmo011, 2021-02-14:
I am trying to communicate via serial a sparkfun pro micro board (with the ATMega 32U4)  with an arduino nano 33 iot. 
I tried with this example for sparkfun pro micro https://pastebin.com/p2i8AzG2

and this example for arduino nano 33 iot https://pastebin.com/MtqyhpLT

but i don't have communication

per1234, 2021-02-14:
Your question is completely off topic for this issue tracker.

This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

Mimmo011, 2021-02-15:
Excuse me for the error!

thanks for link!